ms-iot/vscode-ros #874: why does it export so mang things when I run a launch file？

ku-zhen, 2023-02-11:
hello, every time I run a launch file by this extension, it will export so many things. This will take two seconds, but calling in the terminal does nothing. They don't seem to be doing anything here, how do I remove it?

ubuntu: 20.04 (ros noetic)

[image]

ravijo, 2023-02-11:
I have never seen such a scenario so far. So I need more information. Therefore, can you please explain the steps you followed in detail?

ku-zhen, 2023-02-12:
STEPS:
1. open a catkin workspace with vscode; 
2. use the shortcut key "ctrl+shift+B" to start **catkin_make: build**.   
3. After the build is complete, use the **ROS: Run a ROS launch file(roslaunch)** command. 
4. After selecting the package and launch file , the console will export some things like above before launch the node.

ravijo, 2023-02-13:
@ku-zhen 

Thanks a lot. I noticed the same behavior in my setup. 

By the way, I never use the command palette (Ctrl+Shift+P) inside vscode to run a launch file. Instead, you can type the command directly in terminal inside vscode and that way we don't export those variables. 

Cheers!

polyhobbyist, 2023-03-03:
Launch from roslaunch creates a new terminal; which inherits the vscode launch environment.

ooeygui, 2023-03-27:
Closing as this is by design due to environment inheritance.

closed 2023-03-27 by ooeygui