ms-iot/vscode-ros #878: [bug] Pretty print fix not shipped with v0.8.4?

siferati, 2023-02-22:
- [x] Linux: Ubuntu 20.04.5 LTS
- [x] ROS 1: Noetic

\<Version of the plugin\>
0.8.4
\<Copy the Version information from the `Help | About` menu\>
1.75.1

# what is the bug
- Pretty print is not enabled with ros attach.
- Looks like #815 is not included in v0.8.4?

## Repro steps
1. Attach debugger to running node

# expected behavior
Pretty print is enabled automatically

# additional context
Looking at the tag for v0.8.4 looks like the PR I linked above is not part of the release?
Was that intended? The PR was merged on the same day as the release, so I thought it would be included...
Would be nice to get a new release with the fix so we don't have to manually enable pretty print every time!

polyhobbyist, 2023-03-03:
It looks like it did not ship with 0.8.4. 
0.9.0 has been delayed due to lack of selfhost-hours. There are a couple of core changes which could negatively impact numerous scenarios.

ooeygui, 2023-03-27:
This has been released with 0.9.0. Thank you!

closed 2023-03-27 by ooeygui