ms-iot/vscode-ros #894: [bug]

gsanya, 2023-03-20:
- Linux:  Ubuntu 20.04
- ROS 1: Noetic

\<Version of the plugin\>
0.8.4
\<Copy the Version information from the `Help | About` menu\>
1.74.1

# what is the bug
Warning from roslaunch is taken as error.

## Repro steps
I use an open-source repo where the package naming convention was not followed. There is a package named ORB_SLAM3.
1. When I run the ROS:Launch profile I receive the following error:
Error from roslaunch:
WARNING: Package name "ORB_SLAM3" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits, underscores and dashes.

# expected behavior
roslaunch does not fail on warning.

# additional context
I wouldn't like to change the repo name because it might confuse other users later.

ooeygui, 2023-03-27:
Hi there,
This comes from roslaunch itself, not the ROS extension. This would be a better place to file a bug - https://github.com/ros/ros_comm

WangJitaoNAV, 2024-11-15:
> * Linux:  Ubuntu 20.04
>
>     * ROS 1: Noetic
>
>
> <Version of the plugin> 0.8.4 <Copy the Version information from the `Help | About` menu> 1.74.1
> # what is the bug
>
> Warning from roslaunch is taken as [...]

Hi, I came up with the same issue. Changing the node name in the form of lower case letters is a big work for me. Have you solved the problem?

gsanya, 2024-11-15:
No, sadly that was the only solution I could come up with.

gsanya, 2024-11-18:
> Hi there, This comes from roslaunch itself, not the ROS extension. This would be a better place to file a bug - https://github.com/ros/ros_comm

Sorry for the late reply, but I don't think it's a bug with Roslaunch itself. It should warn me with a warning that I don't use the naming conventions.

The problem is, that the VSCode extension treats this warning as it was an error.

ooeygui, 2024-11-18:
Reopening the issue - we may be interpreting the return result as an error