ms-iot/vscode-ros #908: [bug]

SruthikReddy003, 2023-03-29:
(Please add appropriate labels)
- [ ] Windows: (Version)
- [ ] Linux:  (Dist/Version)
- [ ] ROS 1: Dist
- [ ] ROS 2: Dist

\<Version of the plugin\>
[image]
i have a problem previewing the ROS URDF files in the vs code 
\<Copy the Version information from the `Help | About` menu\>

# what is the bug
\<current behavior\>

## Repro steps
\<how to reproduce the bug. Please attach sample files or github repo if appropriate.\>
1.
2.
3.

# expected behavior
\<what you would expect to happen\>

# additional context
\<any additional information would be helpful for a better understanding\>

v01tt, 2023-03-29:
Hi, I have the same Issue. ROS Preview URDF, renders a completely blank screen. No Grids.
Tried re-installing the ROS plugin, and vscode too. No Luck.
Need help ...

ROS1: Noetic
Linux: 20.04
VScode: 1.76.2
ROS Plugin: 0.9.1

ooeygui, 2023-03-29:
Thanks for reporting. The URDF renderer uses RobotWebTools. Is there a proxy or something in the way?

ooeygui, 2023-03-29:
It looks like RobotWebTools.org is down, which the vscode ROS extension depends on per the recommendation of the docs. Following up with the maintainers.

ooeygui, 2023-03-30:
RobotWebTools.org is down, which is where the pane is sourced. I'm working on contacting the owner to get an ETA for restoration of the website.

I'm also working on a rewrite of the pane to use Babylon.JS which doesn't require an external website - but that's a long lead time item.

hamyyy, 2023-04-15:
Any updates on this? I could really use this feature.

I managed to find most of the js files required [here](https://web.archive.org/web/*/https://static.robotwebtools.org/*)

All are there except Threejs's `ColladaLoader.js`. And most of the files are minified. But that may not pose an issue.
Navigating through wayback machine is a bit of a pain, so I've also attached files here:

[EventEmitter2.min.txt](https://github.com/ms-iot/vscode-ros/files/11239698/EventEmitter2.min.txt)
[ros3d.min.txt](https://github.com/ms-iot/vscode-ros/files/11239699/ros3d.min.txt)
[roslib.min.txt](https://github.com/ms-iot/vscode-ros/files/11239700/roslib.min.txt)
[STLLoader.txt](https://github.com/ms-iot/vscode-ros/files/11239701/STLLoader.txt)
[three.min.txt](https://github.com/ms-iot/vscode-ros/files/11239702/three.min.txt)

Note: I renamed the file extensions to `.txt` so Github will accept them. Just replace `.txt` with `.js`.

You probably don't even need these files as the libraries are still available, its just a matter of finding a new CDN.

[roslib CDN](https://cdnjs.cloudflare.com/ajax/libs/roslibjs/1.1.0/roslib.js)
[ros3d CDN](https://cdn.jsdelivr.net/npm/ros3d@1.1.0/+esm)
[EventEmitter2 CDN](https://cdn.jsdelivr.net/npm/eventemitter2@6.4.9/lib/eventemitter2.min.js)

Maybe you can use these as a temporary fix? Thanks :D

polyhobbyist, 2023-04-17:
Preliminary rewrite in PR - https://github.com/ms-iot/vscode-ros/pull/926. I have to fix a Collada triangle inversion problem in the loader; and fix a CI issue on build.

saddekbouriah, 2023-04-21:
Is it always down ? thank you for your reponse !

ooeygui, 2023-04-21:
the New URDF preview is in main. you can sideload by downloading the vsix from this page (at the bottom) 
https://github.com/ms-iot/vscode-ros/actions/runs/4765688535 and follow one of the methods here - https://stackoverflow.com/questions/42017617/how-to-install-vs-code-extension-manually

NOTE: I'm still testing this, and will likely find bugs. This is almost 100% new code in the last 3 weeks, including a ground up Collada loader...

If you do find a rendering or parsing issue, please file a new bug, attach or point to a URDF which exhibits the issue, and if you can, minimize it to just the blocks that have problems.

Thank you!

saddekbouriah, 2023-04-21:
Thank you , however i have a green grid now but nothing displayed 
PS : it works with rviz

polyhobbyist, 2023-04-21:
@saddekbouriah Can you post a link to the URDF you are testing with? If not the whole thing, can you reduce it to just the part that's broken?

saddekbouriah, 2023-04-21:
yes it's actually a wiki ros tutorial
https://github.com/noshluk2/ros1_wiki/blob/944783c70e6b28c88cf667a1cb9ddc9ccdcd922e/mobile_robot/urdf/rover.urdf.xacro
i have tried with another files but I get the same result + this error Command failed: bash --login -c "xacro '/catkin_ws/src/premier_robot/urdf/test.urdf' && env" mesg: ttyname failed: Inappropriate ioctl for device bash: xacro: command not found 

ps : i have the xacro command installed and it works fine and i have no errors expect this one

Tjaart-Hobby, 2023-04-21:
> 

Hi thanks for the effort, attached 2x screen shots> one green (love the look) with bot display wrong. >the next screen shot from old laptop same file. Also as requested the 
[image]
urdf file used.
[travel_bot.zip](https://github.com/ms-iot/vscode-ros/files/11299733/travel_bot.zip)

[image]

Note. not making use of Xacro's. Once feature are working I will continue build and work out the Xacro's

also noted that 'left-click and drag' or 'right click and drag' and 'wheel-click and drag' > all does the same thing. Could the one pan/tilt/shift view and another move world view?

Tjaart-Hobby, 2023-05-03:
any update please.

ooeygui, 2023-05-03:
@Tjaart-Hobby Thank you for your interest! The URDF previewer is mostly working in the current CI builds - https://github.com/ms-iot/vscode-ros/actions/runs/4812717992, which you can sideload by downloading vscode-ros-dev-vsix. There are a few bugs which I'd like to resolve before pushing to the pre-release pipeline.

ooeygui, 2023-05-04:
v0.9.4 with the new URDF preview has been pushed to Pre-release on the marketplace.  You can select pre-release by going to the marketplace page and select "switch to pre-release"
[image]

ooeygui, 2023-05-04:
Closing this as it the new URDF previewer has been pushed to pre-release.

closed 2023-05-04 by ooeygui

luewh, 2023-08-02:
v0.9.4 pre-release's URDF Preview require xacro to run. Is it avoidable ?
[image]

ooeygui, 2023-08-02:
@luewh Thanks for the feedback. I've been on the fence about usingxacro on urdfs. However, a significant portion of urdfs are actually xacros - including in mainline ROS components such as Nav2 and Turtlebot. 
However, there are many cases where xacro isn't available - such as on the ROS release from Open Robotics or RoboStack. I'm considering a solution which introspects the URDF and uses Xacro only when it detects xacro tags - then warn when Xacro isn't stalled.

ooeygui, 2023-08-02:
@luewh I created this to track this fix - https://github.com/ms-iot/vscode-ros/issues/1084