ms-iot/vscode-ros #910: [bug] ROS1 packages aren't found anymore when using vscode's ROS debugger type (launch file)

rsarrazin2, 2023-03-29:
(Please add appropriate labels)
- [ ] Windows: (Version)
- [x] Linux:  (Ubuntu 20.04 in container)
- [x] ROS 1: noetic
- [ ] ROS 2: Dist

Extension:
0.9.1

vscode:
Version: 1.76.2 (user setup)
Commit: ee2b180d582a7f601fa6ecfdad8d9fd269ab1884
Date: 2023-03-14T17:55:54.936Z (2 wks ago)
Electron: 19.1.11
Chromium: 102.0.5005.196
Node.js: 16.14.2
V8: 10.2.154.26-electron.0
OS: Windows_NT x64 10.0.22621
Sandboxed: Yes

# what is the bug
When launching a ROS1 debug session with a launch file, packages within the workspaces aren't found.

Here the full error message:
```
Command failed: roslaunch --args /foo /tmp/test/src/foo/launch/foo.launch 
Traceback (most recent call last):
  File "/opt/ros/noetic/lib/python3/dist-packages/roslaunch/node_args.py", line 264, in create_local_process_args
    matches = roslib.packages.find_node(node.package, node.type, rospack=rospack)
  File "/opt/ros/noetic/lib/python3/dist-packages/roslib/packages.py", line 417, in find_node
    return find_resource(pkg, node_type, filter_fn=_executable_filter, rospack=rospack)
  File "/opt/ros/noetic/lib/python3/dist-packages/roslib/packages.py", line 519, in find_resource
    pkg_path = rospack.get_path(pkg)
  File "/usr/lib/python3/dist-packages/rospkg/rospack.py", line 207, in get_path
    raise ResourceNotFound(name, ros_paths=self._ros_paths)
rospkg.common.ResourceNotFound: foo
ROS path [0]=/opt/ros/noetic/share/ros
ROS path [1]=/opt/ros/noetic/share

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/opt/ros/noetic/lib/python3/dist-packages/roslaunch/__init__.py", line 268, in main
    node_args.print_node_args(options.node_args, args)
  File "/opt/ros/noetic/lib/python3/dist-packages/roslaunch/node_args.py", line 108, in print_node_args
    args = get_node_args(node_name, roslaunch_files)
  File "/opt/ros/noetic/lib/python3/dist-packages/roslaunch/node_args.py", line 194, in get_node_args
    args = create_local_process_args(node, machine)
  File "/opt/ros/noetic/lib/python3/dist-packages/roslaunch/node_args.py", line 267, in create_local_process_args
    raise NodeParamsException(str(e))
roslaunch.node_args.NodeParamsException: foo
ROS path [0]=/opt/ros/noetic/share/ros
ROS path [1]=/opt/ros/noetic/share
```

## Repro steps
\<how to reproduce the bug. Please attach sample files or github repo if appropriate.\>
1. Check-out [https://github.com/rsarrazin2/catkin_workspace_for_vscode_ros_extension](https://github.com/rsarrazin2/catkin_workspace_for_vscode_ros_extension).
2. Build with `catkin build foo`.
3. Go to the debug activity and launch the "ROS: Launch" configuration.

# expected behavior
The foo package is found and can be debugged.

# additional context
* Reverting to extension version 0.9.0 doesn't solve the problem.
* Reverting to extension version 0.8.4 solves the problem.

ooeygui, 2023-03-29:
If you are blocked by this, you can roll back to 0.8.4 while I investigate.

ooeygui, 2023-03-29:
I'm not sure if this resolves the issue for you, but the launch file is referring to "foo" instead of "foo_node". https://github.com/rsarrazin2/catkin_workspace_for_vscode_ros_extension/blob/189e971d730608f0457cf788ae2cb180a6ce4717/src/foo/launch/foo.launch#L2

It wasn't working for me outside of vscode. I fixed that in the launch file, but still not seeing foo_node propped to the devel directory. 

Still investigating.

ooeygui, 2023-03-29:
I modified your codebase so that it drops to devel, and verified that it runs through vscode.
[catkin_workspace_for_vscode_ros_extension.zip](https://github.com/ms-iot/vscode-ros/files/11104803/catkin_workspace_for_vscode_ros_extension.zip)

ooeygui, 2023-03-29:
Can you try the updated codebase with your scenario?

ooeygui, 2023-03-29:
Can you post your devcontainer descriptor?

rsarrazin2, 2023-03-29:
Thanks for reacting fast on that issue!

Regarding the updated codebase:
* The update can't be started any better with the ROS debugger.
* What is the point of the top-level CMake? Note that I'm using catkin-tools and not catkin-make.

Regarding the devcontainer: I'll create tomorrow morning (European time) a minimum viable example with a working devcontainer based on public images that reproduces (or not!) the problem.

I'll come back and will report.

ooeygui, 2023-03-29:
Every time I press the "go" button, I worry I break something major; thanks for working with me on this.

Ah, catkin-tools...

catkin_make creates the top level cmakefile and the .catkin_workspace file. These are used by the vscode extension to determine if it should behave like a ROS catkin workspace. Catkin-tools does something different.

Thanks for looking into the devcontainer - that will definitely help.

rsarrazin2, 2023-03-30:
I've had a hard time creating a minimum viable devcontainer that reproduces the error. Especially as vscode / extensions modify .vscode/c_cpp_properties.json and .vscode/settings.json. I've tried to reduce the image generation time to a maximum but it still takes several minutes on my system.

Please note also that I'm running vscode on Windows that's connected to a WSL2 Ubuntu 20.04. The devcontainer runs in docker natively installed on the WSL2 Linux, i.e. I don't use Docker Desktop for Windows with its WSL2 backend. It shouldn't matter but I prefer to mention that.

I hope this environment makes it possible on your side to reproduce the issue. From the message error it looks like the ROS debugger 0.9.1 doesn't source ./devel/setup.bash, but it's a pure interpretation of mine.

If there's some extension debug mode I can / shall activate if you can't reproduce on your side, don't hesitate to tell me, I'll try to be as reactive as you!

ooeygui, 2023-03-30:
I'm working on a repro using wsl on 20.04.

ooeygui, 2023-03-30:
I've reordered the setup scripts as part of the fix for #911, which I believe resolved this issue; along with my fixes to the attached workspace.

ooeygui, 2023-03-30:
Could you install this and try it out to ensure it works for your scenario? 
From the CI server- https://github.com/ms-iot/vscode-ros/suites/11925034188/artifacts/624651657

rsarrazin2, 2023-03-31:
> Could you install this and try it out to ensure it works for your scenario? From the CI server- https://github.com/ms-iot/vscode-ros/suites/11925034188/artifacts/624651657

I installed the .vsix file into the devcontainer (using https://stackoverflow.com/a/56055647/1945549) and can confirm the ROS tests are correctly launched. Thanks for the quick fix!

I have to note, though, that now most of my ROS tests launched that way don't pass anymore while they pass when launched manually using `rostest`. I'll dig into it, but in short: The version launched with the ROS debugger stops on a "wait for global callback queue not to be empty anymore" after publishing messages. While this may be an anti-pattern as such, do you see any reason for this behavior change? I'll also have a look at your commits to understand the changes.

ooeygui, 2023-03-31:
@rsarrazin2 Thank you for working with me on this and confirming the fix. I'll push that out today.

on the rostest side - that's a new one. I'll have to debug through that; would it be possible for you to add a test to the "foo" project that reproduces this?

ooeygui, 2023-03-31:
I'm going to close this out now that 0.9.2 has been released. I forked off the rostest issue into https://github.com/ms-iot/vscode-ros/issues/916

closed 2023-03-31 by ooeygui