ms-iot/vscode-ros #911: Configured debug type 'cppvsdbg' is installed but not supported in this environment.

BlueBirdHouse, 2023-03-30:
Suddenly， I am unable to debug the ros2 node.

Configured debug type 'cppvsdbg' is installed but not supported in this environment.

[image]

I have used the VSCode with the ROS plugin for more than one year. Everything works fine. As always, I am happy to upgrade to the latest version. It happened yesterday. All of my programs cannot be debugged anymore. The problem attacks my two separated virtual machines. one is ROS foxy, and the other is ROS humble.

ooeygui, 2023-03-30:
Thank you for reporting. Can you do me a favor and revert back to 0.8.4 to see if it is a C++ debugger issue or a ROS Plugin issue? 

Do you have a sample repo which reproduces this?

ooeygui, 2023-03-30:
Can you also tell me more about your environment? Windows/Ubuntu? Foxy/Humble? Container?

remix-yh, 2023-03-30:
This problem also occurred in my environment.
The conditions under which it occurred are as follows

Ubuntu 22.04
ROS 2 Humble
vscode-ros v0.9.1

When I reverted to v0.8.4, the error no longer occurs.

ooeygui, 2023-03-30:
I can repro. I'm working backwards to see where this came in.

ooeygui, 2023-03-30:
I understand the issue; working on a fix.

ooeygui, 2023-03-30:
Could you install this and try it out to ensure it works for your scenario? 
From the CI server- https://github.com/ms-iot/vscode-ros/suites/11925034188/artifacts/624651657

BlueBirdHouse, 2023-03-31:
Wow, you save the day!
It works fine on all my systems.
Ubuntu 20.04.6 LTS + ROS foxy
Ubuntu 22.04.2 LTS + ROS Humble

ooeygui, 2023-03-31:
Thank you for your help on this! 0.9.2 has been released - closing this out. Don't hesitate to reach out if you have other problems or requests!

closed 2023-03-31 by ooeygui