ms-iot/vscode-ros #916: [bug] ROS1 rostest is failing

ooeygui, 2023-03-31:
Per @rsarrazin2:

I have to note, though, that now most of my ROS tests launched that way don't pass anymore while they pass when launched manually using rostest. I'll dig into it, but in short: The version launched with the ROS debugger stops on a "wait for global callback queue not to be empty anymore" after publishing messages. While this may be an anti-pattern as such, do you see any reason for this behavior change? I'll also have a look at your commits to understand the changes.

rsarrazin2, 2023-04-03:
Interestingly I managed to create a minimal ROS test example... that unexpectedly passes. Going back to my original ROS package, I can't reproduce the problem either. So I suggest to pause / close this issue and I would reopen it if I face the issue again *and* I can reproduce it in a minimal example.

For your reference the [minimal ROS test example](https://github.com/rsarrazin2/catkin_workspace_for_vscode_ros_extension/commit/70c5a76956f4ce65ceeaaa0b5c403393f6f2d0ef) and the [code of concern](https://github.com/rsarrazin2/catkin_workspace_for_vscode_ros_extension/blob/70c5a76956f4ce65ceeaaa0b5c403393f6f2d0ef/src/foo/test/foo_application_test.cpp#L94-L99).

To be started by either:
```bash
catkin test foo
rostest foo foo_application_test.test
```
or using the ROS debugger "foo_application_test" launch.

polyhobbyist, 2023-04-17:
Thank you for sharing this sample. I was trying to reproduce the original issue - but as you noted this no longer happens. I'm going to close this out as you suggested. Thank you!

ooeygui, 2023-04-17:
Closing per conversation. Thank you for working with me on this!

closed 2023-04-17 by ooeygui