ms-iot/vscode-ros #919: [bug]

limshoonkit, 2023-04-06:
(Please add appropriate labels)
- [ ] Windows: (Version)
- [X] Linux:  Ubuntu 20.04 LTS (Focal Fossa) running in WSL2
- [X] ROS 1: Noetic
- [ ] ROS 2: Dist

\<Version of the plugin\>
0.92
\<Copy the Version information from the `Help | About` menu\>

# what is the bug
After updating the extension in visual studio code from 0.84 to 0.92, ROS Core Status no longer display.
[image]

## Repro steps
\<how to reproduce the bug. Please attach sample files or github repo if appropriate.\>
1. Update vscode extension from version 0.84 to 0.92

# expected behavior
\<what you would expect to happen\>
roscore status showing offline status initially.
After running the ROS: Start in vscode command, it will display the published and subscribed topic or services.

Usimian, 2023-04-06:
Just saw this today after updating to 0.9.2
Problem appears on ARM64 native installation, WSL and remote SSH.  All different installations.
'ros2 topic list' shows expected topics

ooeygui, 2023-04-10:
Thanks for the report. I'm currently heads down working on URDF preview which broke due to an external change, then will look at this.

polyhobbyist, 2023-04-17:
Working on this now. I suspect I missed the activation event in package.json

magladko, 2023-07-02:
bug persists for:
vs codeVersion: 1.79.2
OS: Linux x64 5.19.0-46-generic snap (ubuntu 22.04)
extension v0.9.2
ROS2 Humble

Marker at the bottom changes correctly
[image]

But no system status is shown:
[image]

ooeygui, 2023-07-03:
@magladko Thanks for checking in on this. There is a fix in the "pre-release" branch, which you can select from the extension page in VSCode. I'm planning on releasing a new pre-release with many new fixes next week (Week of July 10th), with a push to production shortly afterward.