ms-iot/vscode-ros #938: [bug] Could not render URDF due to: TypeError: Cannot read properties of undefined (reading 'effort') TypeError: Cannot read properties of undefined (reading 'effort')  ...

atticusrussell, 2023-04-21:
(Please add appropriate labels)
- [ ] Windows: (Version)
- [X] Linux:  (Ubuntu 22.04 in WSL2)
- [ ] ROS 1: Dist
- [X] ROS 2: Humble

vscode-ros (sideloaded) v0.9.4

Version: 1.77.0 (user setup)
Commit: 7f329fe6c66b0f86ae1574c2911b681ad5a45d63
Date: 2023-03-29T10:02:16.981Z
Electron: 19.1.11
Chromium: 102.0.5005.196
Node.js: 16.14.2
V8: 10.2.154.26-electron.0
OS: Windows_NT x64 10.0.19044
Sandboxed: No

# what is the bug
Blank URDF preview window opens (with only the green grid floor showing)

Notification:
Could not render URDF due to: TypeError: Cannot read properties of undefined (reading 'effort') TypeError: Cannot read properties of undefined (reading 'effort') at A (https://vscode-remote+wsl-002bubuntu-002d22-002e04.vscode-resource.vscode-cdn.net/home/atticus/.vscode-server/extensions/ms-iot.vscode-ros-0.9.4/dist/webview.js:2:705922) at Module.C (https://vscode-remote+wsl-002bubuntu-002d22-002e04.vscode-resource.vscode-cdn.net/home/atticus/.vscode-server/extensions/ms-iot.vscode-ros-0.9.4/dist/webview.js:2:707129)

## Repro steps
1. clone https://github.com/MSD15440/craft_ros/tree/bf464a1e1f19fdffda3a850b877674d0e48417b8 into a ros workspace source directory
2. open the file craft_ros/craft_description/urdf/craft/urdf.xacro
3. ctrl+shift+p, ROS:Preview URDF

# expected behavior
I would expect the URDF preview to render correctly

polyhobbyist, 2023-04-25:
Fix in progress

ooeygui, 2023-05-04:
A fix for this has been released to the marketplace under the 'prerelease' flag:
[image]

closed 2023-05-04 by ooeygui

kneave, 2023-05-17:
I'm on WSL2/Ubuntu 20.04 with Noetic, I'm seeing this error still on 0.9.3. Is Noetic expected to work or is it only for ROS2 distros please?