ms-iot/vscode-ros #939: [bug] URDF rendering incorrectly with joints misplaced

atticusrussell, 2023-04-21:
(Please add appropriate labels)
- [ ] Windows: (Version)
- [X] Linux:  (Ubuntu 22.04 in WSL2)
- [ ] ROS 1: Dist
- [X] ROS 2: Humble

vscode-ros (sideloaded) v0.9.4

Version: 1.77.0 (user setup)
Commit: 7f329fe6c66b0f86ae1574c2911b681ad5a45d63
Date: 2023-03-29T10:02:16.981Z
Electron: 19.1.11
Chromium: 102.0.5005.196
Node.js: 16.14.2
V8: 10.2.154.26-electron.0
OS: Windows_NT x64 10.0.19044
Sandboxed: No

# what is the bug
[image]

## Repro steps
1. clone https://github.com/joshnewans/articubot_one/tree/8175893af2acdabe3e2ed47601dfc3dca60687d8 into a ros workspace source directory
2. open the file src/articubot_one/description/robot.urdf.xacro
3. ctrl+shift+p, ROS:Preview URDF

# expected behavior
I would expect the URDF preview to render more like this (from gazebo because all URDF previews for old versions are down) 
[image]

# additional context
I think the issue may be that the axes for the joints are swapped in some way.

polyhobbyist, 2023-04-21:
@atticusrussell Thank you!

polyhobbyist, 2023-04-21:
Thank you - I am able to reproduce. Very odd! Debugging now.

polyhobbyist, 2023-04-21:
There are some really good bugs in here.
* Base_footprint is an orphaned tree - I reversed the order of base_link vs base_footprint. Fixed
* I didn't think Cylinder coordinate system mattered, but you offset it to create a smile! Very cute. Fixed.
* STL -> Babylon coordinates are different than Babylon Native mesh coordinate systems; which means my orientation code was wrong.
*

polyhobbyist, 2023-04-25:
Scratching my head on the face... 
[image]

polyhobbyist, 2023-04-25:
(I reduced it, but not quite understanding why it is 90 degrees on the wrong axis)

polyhobbyist, 2023-04-26:
Yea!
[image]

atticusrussell, 2023-04-26:
That looks excellent! Nice work!! I'm excited to try it out whenever the rewrite makes it into a preview/sideload build.

polyhobbyist, 2023-04-26:
I'm working on validating other URDFs to make sure I didn't break anything else before pushing this forward.

ooeygui, 2023-05-04:
A fix for this has been released to the marketplace Pre-release. You can select this by using:
[image]

closed 2023-05-04 by ooeygui