ms-iot/vscode-ros #953: URDF Preview and Roscore Status not working in VS Code 1.78.0

blacksaturn1-wpi, 2023-05-04:
(Please add appropriate labels)
- [X] Windows: Windows 11 Home
- [X] Linux:  WSL 2.0 Ubuntu 20.04
- [X] ROS 1: Noetic

\ROS v.0.9.2

\<Copy the Version information from the `Help | About` menu\>
Version: 1.78.0 (user setup)
Commit: 252e5463d60e63238250799aef7375787f68b4ee
Date: 2023-05-03T20:09:00.748Z
Electron: 22.4.8
Chromium: 108.0.5359.215
Node.js: 16.17.1
V8: 10.8.168.25-electron.0
OS: Windows_NT x64 10.0.22621
Sandboxed: No

# what is the bug
ROS Core Status AND URDF Preview do not work

## Repro steps
1. Launch code from inside terminal, downloads server and starts on Windows
2. Start roscore
3. Double-click ROS1.noetic
4. Shows empty screen for ROS Core Status
5. Click on URDF
6. Ctrl+Shift+P, choose ROS: Preview URDF
7. Blank screen

# expected behavior
1. Show ROS Core status
2. Show URDF preview as seen in RViz

# additional context
\<any additional information would be helpful for a better understanding\>

ooeygui, 2023-05-04:
Thank you for the report! 
A fix has been released to pre-release. You can switch to the reprelease channel by selecting this in the marketplace.
[image]

closed 2023-05-04 by ooeygui

blacksaturn1-wpi, 2023-05-05:
> You don't often get email from ***@***.******@***.***>. Learn why this is important<https://aka.ms/LearnAboutSenderIdentification>
>
> Closed #953<https://github.com/ms-iot/vscode-ros/issues/953> as completed.
>
> -

Thank You, it works!