ms-iot/vscode-ros #1092: Wrong  Coordinate System

G-Yong, 2023-08-10:
(Please add appropriate labels)
- [ ] Windows: (Version)
- [x] Linux:  (Dist/Version)
- [ ] ROS 1: Dist
- [x] ROS 2: Dist

\<Version of the plugin\>
v0.9.4

\<Copy the Version information from the `Help | About` menu\>
[image]

# what is the bug
\<current behavior\>
When I write a urdf file and import it to display in rviz, the displayed model is inconsistent with the result shown in vscode.
[image]

## Repro steps
\<how to reproduce the bug. Please attach sample files or github repo if appropriate.\>
1.Use this urdf file.

`<?xml version="1.0"?>
<robot name="bar_robot">

    
    <link name="base_link">
        <visual>
            <geometry>
                <box size="0.1 0.1 0.1"/>
            </geometry>
            <origin xyz="0 0 0.05" rpy="0 0 0" />
            <material name="blue">
                <color rgba="0 0 1.0 1"/>
            </material>
        </visual>
        <collision>
            <geometry>
                <box size="0.1 0.1 0.1"/>
            </geometry>
            <origin xyz="0 0 0.05" rpy="0 0 0"/>
        </collision>
        
    </link>

    
    <link name="link1">
        <visual>
            <geometry>
                <box size="0.25 0.08 0.02"/>
            </geometry>
            <origin xyz="0.125 0 0.01" rpy="0 0 0" />
            <material name="green">
                <color rgba="0 0.8 0 1"/>
            </material>
        </visual>
        <collision>
            <geometry>
                <box size="0.25 0.08 0.02"/>
            </geometry>
            <origin xyz="0.125 0 0.01" rpy="0 0 0" />
        </collision>
        
    </link>

    
    <joint name="joint1" type="revolute">
        <parent link="base_link"/>
        <child link="link1"/>
        <axis xyz="0 0 1"/>
        
        
        <origin xyz="0 0.05 0.05" rpy="-1.57079 -1.57079 0"/>
        <limit lower="-3.141592653" upper="3.141592653" effort="0" velocity="0"/>
    </joint>

    
    <link name="link2">
        <visual>
            <geometry>
                <box size="0.25 0.08 0.02"/>
            </geometry>
            <origin xyz="0.125 0 0.01" rpy="0 0 0" />
            <material name="red">
                <color rgba="1.0 0 0 1"/>
            </material>
        </visual>
        <collision>
            <geometry>
                <box size="0.25 0.08 0.02"/>
            </geometry>
            <origin xyz="0.125 0 0.01" rpy="0 0 0" />
        </collision>
        
    </link>

    
    <joint name="joint2" type="revolute">
        <parent link="link1"/>
        <child link="link2"/>
        <axis xyz="0 0 1"/>
        <origin xyz="0.25 0 0.02" rpy="0 0 3.141592653" />
        <limit lower="-3.141592653" upper="3.141592653" effort="0" velocity="0" />
    </joint>

    
    <link name="link3">
        <visual>
            <geometry>
                <box size="0.06 0.07 0.05"/>
            </geometry>
            <origin xyz="0.03 0 0.025" rpy="0 0 0" />
            <material name="yellow">
                <color rgba="0.8 0.8 0 1"/>
            </material>
        </visual>
        <collision>
            <geometry>
                <box size="0.06 0.07 0.05"/>
            </geometry>
            <origin xyz="0.03 0 0.025" rpy="0 0 0" />
        </collision>
        
    </link>

    
    <joint name="joint3" type="revolute">
        <parent link="link2"/>
        <child link="link3"/>
        <axis xyz="0 0 1"/>
        <origin xyz="0.22 0 0.02"/>
        <limit lower="-3.141592653" upper="3.141592653" effort="0" velocity="0" />
    </joint>

    
    <link name="link_indicator">
        <visual>
            <geometry>
                <box size="0.05 0.025 0.025"/>
            </geometry>
            <origin xyz="0 0 0.0125" rpy="0 0 0"/>
            
            <material name="yellow">
                <color rgba="0.8 0.8 0 1"/>
            </material>
        </visual>
    </link>
    <joint name="joint_indicator" type="fixed">
        <parent link="link3"/>
        <child link="link_indicator"/>
        <origin xyz="0 0 0.05"/>
    </joint>

</robot>
`

2.To open and preview this URDF file using this plugin in vscode.
3.Somehow to display this urdf using rviz.
4.Comparison of models displayed in vscode and rviz

# expected behavior
\<what you would expect to happen\>
In theory, the coordinates and relationships between links of the urdf model in this plugin should be consistent with the model displayed in rviz.

# additional context
\<any additional information would be helpful for a better understanding\>
I personally think that the reason for this bug may be the incorrect order of rpy, i.e., the execution order of rotation angles for the x, y, and z axes.

ooeygui, 2023-08-30:
I just released 0.9.5 which includes numerous fixes to this codepath. Can you check it out and see if this fixes it for you?

G-Yong, 2023-08-31:
V0.9.5 works.

closed 2023-08-31 by G-Yong