ms-iot/vscode-ros #1138: [bug]

onlytailei, 2023-09-24:
(Please add appropriate labels)
- [x] Linux:  A ubuntu 18.04 melodic docker image.
- [x] ROS 1: melodic

\<Version of the plugin\>  v0.9.5

\<Copy the Version information from the `Help | About` menu\>

# what is the bug
\<current behavior\> 
I run my code in a docker container with the plugin [Dev Container](https://marketplace.visualstudio.com/items?itemName=ms-vscode-remote.remote-containers)

And I set the following environment variables both in .bashrc catkin_ws/devel/setup.bash and .vscode/launch.json                
"ROS_MASTER_URI": "http://172.17.0.2:11311",
"ROS_IP": "172.17.0.2",

I start the roscore in vscode terminal and then start the ROS:launch debug.
However, it always says "Timed out(30 seconds) waiting for ROS Core to start. Start ROSCore manually to avoid this error."

And ros core status always said "offline"

## Repro steps
\<how to reproduce the bug. Please attach sample files or github repo if appropriate.\>
1.
2.
3.

# expected behavior
\<what you would expect to happen\>
run the debug in docker image

# additional context
\<any additional information would be helpful for a better understanding\>

ooeygui, 2024-01-08:
Can you try this with the pre-release version of the extension?

samiframadhan, 2024-04-19:
Hi, i'm also encountering this issue. I have set the debug launch launch.json configuration. However, in my case i have already run the roscore on another container and make the network able to refer to the roscore by setup it in [docker compose config](https://github.com/HowToROS/ros-docker-workspace)

Even though it's not complaining about roscore, it says the debug launch is not supported on ROS 1
[image]

I have already using the pre-release version of vscode-ros

Is there anything i can help to make it happen? Thanks

ooeygui, 2024-04-19:
@samiframadhan Thanks for reaching out. The Launch file debugging feature is designed to enable debugging ROS2 python style launch files, which isn't why it isn't supported on ROS1.

samiframadhan, 2024-04-20:
Oh ok then. Thanks for the confirmation @ooeygui

ooeygui, 2024-04-20:
@samiframadhan That said - this is a real bug. When debugging nodes launched by a launch file, the probe - https://github.com/ms-iot/vscode-ros/blob/43dc595dd8f2c6c410db9f04f25dd26e34e8aee6/src/ros/ros1/ros1.ts#L238 - for roscore  isn't finding the ROS_MASTER_URI correctly. I suspect a race condition loading the environment.

samiframadhan, 2024-04-20:
Oh great, thanks for the pointer. Actually i haven't taken a look at how it works internally. I will try to understand it first and come back when i found something