ms-iot/vscode-ros #1139: [bug] Mirrored models in URDF preview

RangaSundarRS, 2023-09-24:
(Please add appropriate labels)
- [x] Windows: (10 Pro, Version 22H2, OS Build 19045.3448)
- [ ] Linux:  (Dist/Version)
- [ ] ROS 1: Dist
- [x] ROS 2: Humble

\<0.9.5\>

\<1.18.2\>

# what is the bug
When loading the Motoman MH180-120 robot arm, all links' models are mirrored over the link's Y-Z axis. The models are actually in the correct positions, but are displaying flipped.
[image]

## Repro steps
See https://github.com/RangaSundarRS/MotomanMH-ROS2/tree/main for URDF and STL.
1) Source your ROS2 Humble install
2) Build the package with colcon build
3) Source the package with the install\local_setup.ps1
4) Run code to launch VSCode with
5) Open the folder in VSCode.
6) Open urdf\mh180_120.xacro in VSCode.
7) Ctrl-Shift-P -> ROS: Preview URDF
8) View the incorrect display of URDF. Many links are mirrored and rotated around such that the robot arm is no longer even connected!

# expected behavior
The links should appear properly, not mirrored over the Y-Z axis.
Correct appearance (from RoboDK, this is not using the URDF):
[image]

Correct visualization of the URDF (From urdf-viz):
[image]

ooeygui, 2023-09-27:
Thank you - invesetigating.

AVJF1996, 2023-12-01:
I have the same issue
[image]

ooeygui, 2023-12-04:
Oops, that's embarrassing. Thanks for bringing it to my attention

jweaston, 2024-06-11:
I am also experiencing the same issue with the Fanuc CRX10ial.
[image]
Let me know if there's any other information I can provide to help debug this issue.

polyhobbyist, 2024-06-27:
This was a "handed coordinate system" problem. Corrected by inverting the meshes. I'm testing to make sure I didn't regress anything else.
[image]

polyhobbyist, 2024-06-27:
I also noticed that the materials aren't applied by name; for example `<material name="yellow"/>` doesn't render because the RGBA isn't included.

polyhobbyist, 2024-06-27:
@RangaSundarRS Do you mind if I include a flattened version of your xacro in the test case for the rendering?

Usimian, 2024-07-21:
Has this been fixed?  URDF preview in vscode is very useful.
For now I've gone back to urdf-viz which renders properly.