ms-iot/vscode-ros #1150: [bug] Debugger unrecognized remap tag in <include> tag

sandilyasg, 2023-09-27:
(Please add appropriate labels)

- [ ] Linux:  (Ubuntu 18.04)
- [ ] ROS 1: Melodic

\<Version of the plugin\>
v0.8.4

\<Copy the Version information from the `Help | About` menu\>
Version: 1.82.2
Commit: abd2f3db4bdb28f9e95536dfa84d8479f1eb312d
Date: 2023-09-14T05:51:20.981Z
Electron: 25.8.1
ElectronBuildId: 23779380
Chromium: 114.0.5735.289
Node.js: 18.15.0
V8: 11.4.183.29-electron.0
OS: Linux x64 5.4.0-150-generic snap

# what is the bug
\<current behavior\>
I get the following error from VS code debugger: "Error from roslaunch: WARN: unrecognized 'remap' tag in tag"

## Repro steps
\<how to reproduce the bug. Please attach sample files or github repo if appropriate.\>
1. I have 2 main launch files, one that I launch from the linux terminal, and one via the launch.json in VS Code (both shown below).
Launched in VS code: master_rag_simulation.launch:
`<launch>

<include file="$(find turtlebot3_slam)/launch/multi_karto_gtsam_posfilter.launch"/>

<include file="$(find resource_aware_coordination)/launch/multi_local_comm_nodes.launch"/>

</launch>`

Launched from terminal: master_rag_sim_terminal.launch:

`<launch>

<include file="$(find turtlebot3_gazebo)/launch/multi_MAIN_turtlebot3_testzone.launch"/>

<include file="$(find turtlebot3_slam)/launch/multi_posfilter.launch"/>

<include file="$(find turtlebot3_navigation)/launch/multi_move_base.launch"/>

<include file="$(find resource_aware_coordination)/launch/exploration_frontier_points.launch"/>

</launch>`
When I launch this file, particularly the "multi_karto_gtsam_posfilter.launch" file, I get the following error from VS code:

"Error from roslaunch: WARN: unrecognized 'remap' tag in tag"

This is what my multi_karto_gtsam_posfilter.launch file looks like:
`<launch>
  <arg name="open_rviz" default="true"/>
  <param name="/use_sim_time" value="true"/>
  <arg name="output_to_screen" default="screen"/> 

  
  

    <include file="$(find turtlebot3_slam)/launch/turtlebot3_karto_gtsam.launch">
      
      <arg name="output_to_screen" value="screen"/>
      <arg name="ns" value="tb3_0"/>
      <remap from="/$(arg ns)/scan" to="/$(arg ns)/scan_filtered"/>
    </include>

  
    <include file="$(find turtlebot3_slam)/launch/turtlebot3_karto_gtsam.launch">
      <arg name="output_to_screen" value="log"/>
      <arg name="ns" value="tb3_1"/>
      <remap from="/$(arg ns)/scan" to="/$(arg ns)/scan_filtered"/>
    </include>

  
    <include file="$(find turtlebot3_slam)/launch/turtlebot3_karto_gtsam.launch">
      <arg name="output_to_screen" value="log"/>
      <arg name="ns" value="tb3_2"/>
      <remap from="/$(arg ns)/scan" to="/$(arg ns)/scan_filtered"/>
    </include>

  <include file="$(find turtlebot3_gazebo)/launch/multi_map_merge_karto.launch"></include>

  <group if="$(arg open_rviz)"> 
    <node pkg="rviz" type="rviz" name="rviz" required="true"
          args="-d $(find turtlebot3_gazebo)/rviz/multi_turtlebot3_slam_withpathspose.rviz"/>
  </group>

</launch>`

# expected behavior
\<what you would expect to happen\>
I expect to see no error or warning, and the debugger steps through breakpoints in the C++ nodes.

# additional context
\<any additional information would be helpful for a better understanding\>
I see that this error originates from [tools/roslaunch/src/roslaunch/xmlloader.py](https://github.com/strawlab/ros_comm/blob/master/tools/roslaunch/src/roslaunch/xmlloader.py), line 583. When I run the launch files from my terminal, they work correctly with no errors.

xoofee, 2023-10-10:
I met the same problem with python launch file. Ros extension in VSCODE ignore the remappings  in launch.py when start debuging

If run command in terminal, it has no problem

sandilyasg, 2023-10-10:
@xoofee what do you mean Python launch file? As in you have a .launch file which includes a python node? Yes if run from the terminal, this remapping problem does not occur. Would be nice to get it to work in VS code ROS extension

xoofee, 2023-10-10:
yes, I mean xxx.launch.py file in python format, this is an example

`from launch import LaunchDescription
from launch_ros.actions import Node

def generate_launch_description():
    return LaunchDescription([
        Node(
            package='ros_test',
            namespace='',
            executable='ros_test_node',
            remappings=[
                ('~/input', '/localization/kinematic_state'),
            ]            
        ),
    ])
`

However, if I do not use the tilde ~/ in my topic name, just 'input', the VS code ROS extension has no problem for the remap.

sandilyasg, 2023-10-10:
@xoofee thanks for your reply. Where do I find this .launch.py file, is it generated for any .launch file that I have or did you specifically create it yourself for use with VSCode ROS extension?

ooeygui, 2024-01-08:
Just to confirm -This looks like it repros on both ROS1 and ROS2?

CN-Antonio, 2025-02-06:
I have the same problem with ROS2 Humble which I'm currently using.

My launch file is xml form. Same as mentioned above, if remap from topics with "~", the remapping will be ignored. The actual topic name is the same as that in source file.

My vscode launch.json is like:
```json
      {
            "name": "ROS: Launch",
            "type": "ros",
            "request": "launch",
            "target": "${workspaceFolder}/install/{Path to launch.xml}",
        },
```

My ROS launch.xml file is like
```xml
<arg name="input/cam_front_left" default="/CAM_FRONT_LEFT/image_raw"/>

<node name="pkg_node" pkg="pkg_ros" exec="pkg_ros_node" output="screen" >
    <param from="$(var ros_param_path)" allow_substs="true"/>

    <remap from="~/input/image_fl" to="$(var input/cam_front_left)"/>
</node>
```

When run launch in console with command `ros2 launch package launch.xml`, it's confirmed that the remapping is correct. There's a topic named "/CAM_FRONT_LEFT/image_raw" with ros2 topic list check.

But when run debug in vscode, the remapping is ignored. Only the topic "/pkg_node/input/image_fl" is shown.

Still run debug in vscode, but change the remap as below
```xml
<!-- use specified pkg_node to replace ~ -->
<remap from="pkg_node/input/image_fl" to="$(var input/cam_front_left)"/>
```
The remap back to work.

So it may not to be ROS's bug.