ms-iot/vscode-ros #1163: [bug] unable to initialize ROS extension when in a ROS2 workspace

machutaSICKAG, 2023-10-09:
(Please add appropriate labels)
- [x] Windows: (21H2 (Build 19044.3448)
- [x] Linux:  (WSL2, Ubuntu 22.04)
- [x] ROS 2: Dist

\<Version of the plugin\>
0.9.2

\<Copy the Version information from the `Help | About` menu\>
Version: 1.83.0 (user setup)
Commit: e7e037083ff4455cf320e344325dacb480062c3c
Date: 2023-10-03T16:12:16.321Z
Electron: 25.8.4
ElectronBuildId: 24154031
Chromium: 114.0.5735.289
Node.js: 18.15.0
V8: 11.4.183.29-electron.0
OS: Windows_NT x64 10.0.19044

# what is the bug
\<current behavior\>
Currently, whenever I launch VSCode from WSL2 within a ROS2 workspace (either from the source folder or from the main workspace folder), the notification "ROS Extension Initializing" pops up and doesn't go. The ROS version is then not recognized.

When VSCode is launched from any other location outside of the workspace, the extension initialization goes without a problem, and a correct version of ROS2 is displayed in the status bar (Humble).

# expected behavior
\<what you would expect to happen\>
Correct initialization of the ROS2 workspace. It used to work.

ooeygui, 2024-01-08:
Thanks for the report. I appreciate that you added the zsh context.

Can you try the pre-release version of the extension? (just to rule it out, I've made major modifications to the startup)

elkuno213, 2024-03-29:
I think this bug hasn't been fixed yet, even using the pre-release version. I  found the ROS extension always ignores the option `ros.rosSetupScript` and sources `setup.bash` by default.

**Reproduction**
---

Here is how I reproduced the problem (my workspace is `/workspaces/ros2-ws`):

`.vscode/settings.json`
```json
{
  "ros.distro": "iron",
  "ros.rosSetupScript": "${workspaceFolder}/install/local_setup.zsh",
}
````

The output shows that the ROS extension tries to source 3 times for each `setup.bash`:

```
Sourcing Environment using: bash --login -c "source '/opt/ros/iron/setup.bash' && env"
Sourcing Environment using: bash --login -c "source '/workspaces/ros2-ws/install/setup.bash' && env"
Sourcing Environment using: bash --login -c "source '/opt/ros/iron/setup.bash' && env"
Sourcing Environment using: bash --login -c "source '/workspaces/ros2-ws/install/setup.bash' && env"
Sourcing Environment using: bash --login -c "source '/opt/ros/iron/setup.bash' && env"
Sourcing Environment using: bash --login -c "source '/workspaces/ros2-ws/install/setup.bash' && env"
```

It doesn't matter the setup file is `local_install.zsh`, `install.zsh` or `local_install.bash`, the extension forces to source `install.bash`

**Remarks**
---

I found quite difficult to debug using this extension when creating new package because the setup script is not updated. My temporary solution is to try to comment the option `ros.rosSetupScript` or reload VSCode window ; so that the ROS extension restarts the sourcing action. 

I suggest to have a command, so called `updateSetupScript` to make this process easier. :) Many thanks!

ooeygui, 2024-03-29:
Thank you for reporting. This extension does expect the default shell to be bash; it needs to be updated to respect the user's shell.