ms-iot/vscode-ros #1197: [bug] The debugger can't connect to the source file

ostepan2006, 2023-11-04:
(Please add appropriate labels)
- [ ] Windows: (Version)
- [ +] Linux:  (Ubuntu 18.04)
- [ +] ROS 1: Dist
- [ ] ROS 2: Dist

ros extention 0.9.2.

vscode
Version: 1.84.0
Commit: d037ac076cee195194f93ce6fe2bdfe2969cc82d
Date: 2023-11-01T11:28:21.782Z
Electron: 25.9.2
ElectronBuildId: 24603566
Chromium: 114.0.5735.289
Node.js: 18.15.0
V8: 11.4.183.29-electron.0
OS: Linux x64 5.4.0-150-generic snap

# what is the bug
\<current behavior\>
The debugger is not connected with source code.
Breakpoints are not working and have gray color (see picture below).

[image]

## Repro steps
\<how to reproduce the bug. Please attach sample files or github repo if appropriate.\>
1. Run catkin make -DCMAKE_BUILD_TYPE=Debug
2. Run a launch file with svo_ros node.
3. 

# expected behavior
\<what you would expect to happen\>
The debugger should been connected to the source file.  
The breakpoint should been hit.

# additional context
\<any additional information would be helpful for a better understanding\>
Debugging of toy example (publisher.cpp) had been working.

ooeygui, 2024-01-08:
Could you try this with the pre-release version?

Achllle, 2025-01-08:
Does not work with the pre-release version

ooeygui, 2025-01-09:
I'm working on splitting the extension into a ROS 1 and ROS 2 versions. I'll ensure that debugging is functioning in that new release.