ms-iot/vscode-ros #1224: Collision URDF and Xacro Visulizer[feature]

AVJF1996, 2023-11-27:
Hi there

**What is needed**
It's a good idea to add a check box to show the collision URDF visualizer, I think it's not hard to add this feature.

**Why is it needed**
collision properties is more vital to visualize than visual tag However most of the time they are the same.

ooeygui, 2024-01-08:
I love this!

ArsenicBismuth, 2024-02-05:
Man this feature would be great. My robot has mesh-based visualization and primitives collision (to simplify collision detection). Right now editing it without any visualization is highly time-consuming task.

So, was sad to discover this is still a new feature request haha.

ooeygui, 2024-02-05:
@ArsenicBismuth Thanks for the feedback! I'm glad you like the idea of it - I wish I had the 'human energy' to complete this sooner. It is next on my todo list though.