ms-iot/vscode-ros #1240: breakpoints not working !

ZhaoJunnNie, 2023-12-14:
(Please add appropriate labels)
- [ ] Windows: (Version)
- [x] Linux:  (Dist/Version) Ubuntu20.04
- [x] ROS 1: Dist noetic
- [ ] ROS 2: Dist

\<Version of the plugin\>
v0.9.2
\<Copy the Version information from the `Help | About` menu\>

# what is the bug
\<current behavior\>
use the ROS Launch in  Debug, but breakpoints are not working, launch file has just one node 
## Repro steps
\<how to reproduce the bug. Please attach sample files or github repo if appropriate.\>
1.git clone https://github.com/shichaoy/cube_slam.git 
2.install necessary packages
3.catkin_make -DCMAKE_BUILD_TYPE=Debug
4.create Debug for ROS Launch in "[object_slam](https://github.com/shichaoy/cube_slam/tree/master/object_slam)" and "[object_slam.launch](https://github.com/shichaoy/cube_slam/blob/master/object_slam/launch/object_slam.launch)" which comments rviz node

# expected behavior
\<what you would expect to happen\>
programs stop before breakpoints 
# additional context
I have tried to rm build and devel directory and catkin_make -DCMAKE_BUILD_TYPE=Debug, but it is still not working , here are some files in use 
[image]
[image]
[image]
[image]
[image]

ZhaoJunnNie, 2023-12-14:
Sorry, I made a mistake on infos. code is on  Linux: (Dist/Version) Ubuntu16.04  ROS 1: Dist kinetic, it is a docker container on Ubuntu20.04

ooeygui, 2024-01-08:
Hi there, 
Thank you for the reply. 16.04 is not officially supported.  
Can you switch to pre-release in the container and try again?

ZhaoJunnNie, 2024-01-13:
Thanks for your reply! I finished the problem by adding "-DCMAKE_CXX_FLAGS=-g " in task.json without switching to pre-release