ms-iot/vscode-ros #1269: [bug] ROS2 breakpoints not working !

jalenzz, 2024-01-10:
- [X] Linux: Ubuntu 22.04
- [X] ROS 2: humble

\<Version of the plugin\>
v0.9.2

\<Copy the Version information from the `Help | About` menu\>
Version: 1.85.1
Commit: 0ee08df0cf4527e40edc9aa28f4b5bd38bbff2b2
Date: 2023-12-13T09:47:11.635Z
Electron: 25.9.7
ElectronBuildId: 25551756
Chromium: 114.0.5735.289
Node.js: 18.15.0
V8: 11.4.183.29-electron.0
OS: Linux x64 6.2.0-39-generic

# what is the bug
\<current behavior\>

I'm trying to debug a ros2 program that was started via a launch file, and I hit a breakpoint
When I turn on debugging, the program runs normally, but it doesn't stop at the breakpoint.

```json
// launch.json
{
    "version": "0.2.0",
    "configurations": [
        {
            "name": "ROS: Launch",
            "request": "launch",
            "target": "path to launch",
            "type": "ros",
            "preLaunchTask": "debug workspace"
        }
    ]
}
```
```json
// a part of tasks.json
        {
            "label": "debug workspace",
            "detail": "Build ROS 2 workspace with debug symbols",
            "type": "shell",
            "command": "colcon build --symlink-install  --cmake-args '-DCMAKE_BUILD_TYPE=Debug'",
            "group": "build",
            "problemMatcher": "$gcc"
        },

```

# additional context
\<any additional information would be helpful for a better understanding\>
My ros2 program is written in components, and the launch in the launch file is through ComposableNodeContainer. Will this affect the debug?
```py
    serial = ComposableNodeContainer(
        name='serial',
        namespace='',
        package='rclcpp_components',
        executable='component_container',
        composable_node_descriptions=[
            ComposableNode(
                package = 'serial',
                plugin = 'sensor::SerialForUnity',
                name = 'serial_for_unity_node',
                parameters = [config]
            )
        ]
    )
```

Ryanf55, 2024-01-13:
I have this problem too. I switched to pre-release to check. 

**launch.json**
```json
        {
            "name": "ROS: Launch",
            "type": "ros",
            "request": "launch",
            "target": "/home/ryan/Dev/ros2_ws/src/grid_map_geo/install/grid_map_geo/share/grid_map_geo/launch/load_vrt_launch.xml",
            "preLaunchTask": "colcon build debug"
        }
```

**tasks.json**
```json
        {
            "label": "colcon build debug",
            "detail": "Build ROS 2 workspace with debug symbols",
            "type": "shell",
            "command": "colcon build --symlink-install --mixin debug",
            "group": "build",
            "problemMatcher": "$gcc"
        },
```

You can see that the debug panel knows about the breakpoints, but they aren't hit. 
[image]

jalenzz, 2024-01-15:
Yes, I had tried pre-release version, but still not work.

Is you used composable nodes?

I tried normal nodes, and they stop at breakpoints, but not composable nodes.

jalenzz, 2024-01-15:
```py
from launch import LaunchDescription
from launch_ros.actions import ComposableNodeContainer
from launch_ros.descriptions import ComposableNode

def generate_launch_description():

    composable_node_container = ComposableNodeContainer (
        name='composable_node_container',
        namespace='',
        package='rclcpp_components',
        executable='component_container',
        composable_node_descriptions=[
            ComposableNode(
                package = 'camerainfo',
                plugin = 'camerainfo::CameraInfoNode',
                name = 'camera_info_node',
                extra_arguments = [{"use_intra_process_comms": True}],
            ),
            ComposableNode(
                package = 'camera',
                plugin = 'sensor::CameraNode',
                name = 'camera_node',
                extra_arguments = [{"use_intra_process_comms": True}]
            )
        ]
    )
    return LaunchDescription([composable_node_container])
```

Ryanf55, 2024-01-15:
> @Ryanf55 Yes, I had tried pre-release version, but still not work.
> 
> Is you used composable nodes?
> 
> I tried normal nodes, and they stop at breakpoints, but not composable nodes.

Only normal nodes were attempted. I can't get any of it to work now.