ms-iot/vscode-ros #1271: [bug] rosrun export error

Moon-404, 2024-01-15:
(Please add appropriate labels)
- [x] Windows: (11)
- [x] Linux:  (wsl ubuntu 20.04)
- [x] ROS 1: neotic

\<Version of the plugin\>
0.9.2

\<Copy the Version information from the `Help | About` menu\>
Version: 1.85.1 (user setup)
Commit: 0ee08df0cf4527e40edc9aa28f4b5bd38bbff2b2
Date: 2023-12-13T09:49:37.021Z
Electron: 25.9.7
ElectronBuildId: 25551756
Chromium: 114.0.5735.289
Node.js: 18.15.0
V8: 11.4.183.29-electron.0
OS: Windows_NT x64 10.0.22621

# what is the bug
\<current behavior\>
When using `>ROS: Run` in vscode, the terminal exports LS_COLORS but get errors like:
```
35:do=01: command not found
35:bd=40: command not found
33: command not found
01:cd=40: command not found
33: command not found
01:or=40: command not found
31: command not found
01:mi=00:su=37: command not found
```

## Repro steps
\<how to reproduce the bug. Please attach sample files or github repo if appropriate.\>
1. open a catkin workspace with vscode;
2. use the shortcut key "ctrl+shift+B" to start catkin_make: build.
3. After the build is complete, use the rosrun command.

# expected behavior
\<what you would expect to happen\>
no export errors

# additional context
\<any additional information would be helpful for a better understanding\>
It seems terminal inherit problem, but there's some problems about LS_COLORS
The issue is similar: https://github.com/ms-iot/vscode-ros/issues/874

ooeygui, 2024-01-15:
Can you try the pre-release version of the extension?
I'll look into this.

Moon-404, 2024-01-16:
> Can you try the pre-release version of the extension? I'll look into this.

still the same problem