ms-iot/vscode-ros #1272: [bug] ROS distro not sourced for tasks.json on humble

Ryanf55, 2024-01-15:
(Please add appropriate labels)
- [ ] Windows: (Version)
- [x] Linux:  (Dist/Version) Ubuntu 22.04 for development server, Windows 10 for host OS, connected using vscode ssh extension
- [ ] ROS 1: Dist
- [ ] ROS 2: ROS 2 humble

Plugin v0.9.6 pre-release

[image]

# what is the bug

When I add a task to run colcon build, it does not source the ROS environment.

## Repro steps
1. Add `"ros.distro": "humble",` to .vscode/settings.json on the remote computer
2. Add this to tasks.json
    ```json
           {
            "label": "colconBuildDebug",
            "detail": "Build ROS 2 workspace with debug symbols",
            "type": "shell",
            "command": "colcon build --symlink-install --mixin debug",
            "group": "build",
            "problemMatcher": "$gcc"
        },
    ```
3. Ensure the ROS extension is started and running in pre-release on the development server
4. Type F1: `Tasks: Run task` and select `colconBuildDebug`
5. Observe the output showing that it never sourced the ROS environment
```
 *  Executing task: colcon build --symlink-install --mixin debug 

Starting >>> grid_map_geo
--- stderr: grid_map_geo                         
Traceback (most recent call last):
  File "/opt/ros/humble/share/ament_cmake_core/cmake/package_templates/templates_2_cmake.py", line 21, in <module>
    from ament_package.templates import get_environment_hook_template_path
ModuleNotFoundError: No module named 'ament_package'
CMake Error at /opt/ros/humble/share/ament_cmake_core/cmake/ament_cmake_package_templates-extras.cmake:41 (message):
  execute_process(/usr/bin/python3.10
  /opt/ros/humble/share/ament_cmake_core/cmake/package_templates/templates_2_cmake.py
  /home/ryan/Dev/ros2_ws/src/grid_map_geo/build/grid_map_geo/ament_cmake_package_templates/templates.cmake)
  returned error code 1
Call Stack (most recent call first):
  /opt/ros/humble/share/ament_cmake_core/cmake/ament_cmake_coreConfig.cmake:41 (include)
  /opt/ros/humble/share/ament_cmake/cmake/ament_cmake_export_dependencies-extras.cmake:15 (find_package)
  /opt/ros/humble/share/ament_cmake/cmake/ament_cmakeConfig.cmake:41 (include)
  CMakeLists.txt:18 (find_package)

---
Failed   <<< grid_map_geo [0.13s, exited with code 1]

Summary: 0 packages finished [0.50s]
  1 package failed: grid_map_geo
  1 package had stderr output: grid_map_geo

 *  The terminal process "/bin/bash '-c', 'colcon build --symlink-install --mixin debug'" terminated with exit code: 1. 
 *  Terminal will be reused by tasks, press any key to close it. 
 ```

# expected behavior

Colcon build will be successful and first `source /opt/ros/humble/setup.bash` before running.

# additional context

Even if I add `source /opt/ros/humble/setup.bash` to my bashrc, it doesn't source.

As a workaround, I could do modify the task command like so, but that is not documented:
            `"command": "source /opt/ros/humble/setup.bash && colcon build --symlink-install --mixin debug",`
            
  Then, the launch still fails because it's not sourcing the local ROS workspace before launch:
  
[image]

Ryanf55, 2024-01-15:
How is launch.json supposed to work if it cannot source the local environment?

ooeygui, 2024-01-15:
The problem here is using Task.json; I don't believe there is a hook for the ROS extension in that codepath of VSCode to influence the *task* environment. 

We register a VSCode launch.json handler which is ROS aware - and is able to source the environment. The Task..json file is not used when you use the ROS mechanism, so the extension can control the environment.

ooeygui, 2024-01-16:
The VSCode ROS extension Provides a "ROS Create Terminal" Command which creates a new terminal with ROS sourced. VSCode does not give extensions a hook for providing environment through the normal UI.

Ryanf55, 2024-01-20:
Hello,

I'm not sure if you need more info, but it appears this bug is only present when SSH'ing from Windows. If I have direct access to the Linux computer, the launch task works fine, and colcon build tasks works fine with the workaround I posted.

It would be desirable for SSH not to adversely impact this extension's functionality.

ooeygui, 2024-01-21:
@Ryanf55 That interesting - thanks for the note. It certainly narrows down where this could be going wrong.