ms-iot/vscode-ros #1291: [bug] Debug not stopping at breakpoints in ROS2 (python) launch file

slim71, 2024-05-10:
## System info
- [ ] Windows: (Version)
- [x ] Linux: 22.04
- [ ] ROS 1: Dist
- [x ] ROS 2: Humble

## Version of the plugin
v0.8.6 (pre-release)

## Version information from the `Help | About` menu
Version: 1.89.1
Commit: dc96b837cf6bb4af9cd736aa3af08cf8279f7685
Date: 2024-05-07T05:16:23.416Z
Electron: 28.2.8
ElectronBuildId: 27744544
Chromium: 120.0.6099.291
Node.js: 18.18.2
V8: 12.0.267.19-electron.0
OS: Linux x64 6.5.0-28-generic snap

# What is the bug
When using the functionality defined in launch.json, the launch file seems to be executed (terminals are created for each node spawned) but the breakpoints in the launch file itself are ignored.

## Repro steps
1. Build the launch package
2. Execute the launch file desired

# Expected behavior
The debugger should stop at the breakpoint inserted in the launch file.

# Additional context
The command I'm trying to use from the `launch.json` file is
```
{
      "name": "ROS: Launch File (isolated-install)",
      "type": "ros",
      "request": "launch",
      "preLaunchTask": "debug_build",
      "target": "${workspaceFolder}/install/${input:package}/share/${input:package}/launch/${input:ros_launch}.launch.py",
    }
```
and the `debug_build` task is defined as
```
        {
            "label": "sourceros",
            "detail": "Source the ROS2 global workspace",
            "type": "shell",
            "command": "source /opt/ros/humble/setup.bash",
            "problemMatcher": "$gcc"
        },
        {
            "label": "sourceall",
            "detail": "Source the local workspace",
            "type": "shell",
            "command": "source install/setup.bash",
            "problemMatcher": "$gcc",
            "dependsOrder": "sequence",
            "dependsOn": [
                "sourceros"
            ]
        },
        {
            "label": "debug_build",
            "detail": "Build the specified package in debug mode",
            "type": "shell",
            "command": "colcon build --packages-select ${input:package} --cmake-args '-DCMAKE_BUILD_TYPE=Debug' -Wall -Wextra -Wpedantic -DCMAKE_CXX_FLAGS=-g -DDEBUG_MODE=1",
            "problemMatcher": "$gcc",
            "dependsOrder": "sequence",
            "dependsOn": [
                "sourceros",
                "sourceall"
            ]
        }
```
The arguments provided to the `colcon build` command is the complete list of flags I've ended up trying! But still, since it is a python launch package, I'm not sure it should work...

jweaston, 2024-06-11:
You need to run this launch description to debug launch files specifically.

`    {
      "name": "ROS: Debug Launch File (merge-install)",
      "type": "ros",
      "request": "debug_launch",
      "target": "${workspaceFolder}/install/share/${input:package}/launch/${input:ros_launch}"
    }`

slim71, 2024-06-11:
Thank you! 😁 
Following your input, I've managed to debug the launch file itself, even in isolated-install! I've used
```json
{
      "name": "ROS: Debug Launch File (isolated-install)",
      "type": "ros",
      "request": "debug_launch",
      "target": "${workspaceFolder}/install/${input:package}/share/${input:package}/launch/${input:ros_launch}.launch.py"
}
```
I'll close this then, but I would like further feedback if possible.

Based on what I've encountered, what changes the debugging behavior is the request then, right?
The one I had before has the `launch` request, which results in using the launch files only to launch nodes and debugging is done on those nodes.
The correct one to debug the actual launch file is the one with the `debug_launch` request, as you told me.
Is that right? Also, did I miss docs on this by any chance?
Thank you very much! 😄

closed 2024-06-11 by slim71

jweaston, 2024-06-12:
I'll be honest I don't remember where I found this out but yes the `request` field is how you change between the launch file and the actual nodes.

slim71, 2024-06-13:
ehi, that's good enough for me! 😝 
thank you!

slim71, 2025-01-13:
I can't remember exactly, but I think I had to put breakpoints not in the "active" source code, but in the built ones in the `install` folder...
Also, are you sure you built with debug info?