ms-iot/vscode-ros #1296: [bug] ROS2 launch processor was unable ro produce a node list.

gsanya, 2024-08-07:
(Please add appropriate labels)
- [ ] Windows: (Version)
- [x] Linux: Ubuntu 22.04 (Dist/Version)
- [ ] ROS 1: Dist
- [x] ROS 2: Humble

\<Version of the plugin\> 
v0.9.2

\<Copy the Version information from the `Help | About` menu\>
Version: 1.92.0
Commit: b1c0a14de1414fcdaa400695b4db1c0799bc3124
Date: 2024-07-31T23:26:45.634Z
Electron: 30.1.2
ElectronBuildId: 9870757
Chromium: 124.0.6367.243
Node.js: 20.14.0
V8: 12.4.254.20-electron.0
OS: Linux x64 6.5.0-44-generic snap

# what is the bug

\<current behavior\>
The following launch.json file doesn't start the debugging:
```yaml
{
    "configurations": [
    {
        "name": "ROS: Launch",
        "type": "ros",
        "request": "launch",
        "target": "/home/appuser/spot_ros2/src/spot_ros2/spot_driver/launch/rviz.launch.py",
        "arguments": ["config_file:=/home/appuser/spot_ros2/src/spot_ros2/spot_driver/config/our_spotty.yaml",
                    "spot_name:=spotty publish_point_clouds:=True",
                    "launch_rviz:=True",
                    "uncompress_images:=False",
                    "publish_compressed_images:=True"]
    }    
    ]
}
```
It produces the following error popup:
[image]

While the following 2 commands start the launch file correctly (so it's not a problem with the launch script):
```bash
source install/setup.bash
ros2 launch spot_driver spot_driver.launch.py config_file:=/home/appuser/spot_ros2/src/spot_ros2/spot_driver/config/our_spotty.yaml spot_name:=spotty publish_point_clouds:=True launch_rviz:=True uncompress_images:=False publish_compressed_images:=True
```

## Repro steps
\<how to reproduce the bug. Please attach sample files or github repo if appropriate.\>
1. Create the launch file above
2. ctrl+shift+p: ROS: Start
3. On the run and debug pane click the green arrow beside ROS: Launch

# expected behavior
\<what you would expect to happen\>
The debugger starts the nodes are debugable.

# additional context
\<any additional information would be helpful for a better understanding\>
This shouldn't cause a problem, but I try to make the debugger work inside a dev_container.

gsanya, 2024-08-09:
I don't fully understand the problem yet, but managed to solve it with a little hacking....

The problematic launch file is this one:
https://github.com/bdaiinstitute/spot_ros2/blob/main/spot_driver/launch/spot_driver.launch.py

It has the `generate_launch_description` function which returns the launch description. But the executable nodes are made in the `launch_setup` function which is added as an opaque function to the LaunchDescriptor.

When VSCode starts the Debuggers it processes the launch files with the following script: https://github.com/ms-iot/vscode-ros/blob/master/assets/scripts/ros2_launch_dumper.py

Which uses a different method to get all nodes than the original launch package in the ros2 og repo and this difference is causes the problem.

It iterates over the launch descriptor and if any entity adds a new entity its added to the list, so it will be iterated over. The problem is in line 112 where the visit function is called. If it returns with some entity it is added to the list, but if it returns with none, nothing is added. 

In this particular launch file the opaque function adds new entities to the original LanchDescriptor and the visit function returns None. This is problematic, because the original LaunchDescriptor is not checked so the entities are not added to the list, and thus the executable nodes are never started.

I rewrote the `launch_setup` function so it returns the entity list, and thus the visit function returns with not None. The original ros2 launch processor can deal with this so I think some checks should be added to see if something is added to the LaunchDescriptors entity list later on in the launch process.

Scoeerg, 2025-06-05:
FYI:

I got the same error `ros2 launch processor was unable to produce a node list`, but the solution was to make the sym-link installed python script executable. 

**tl;dr:** 

```bash
chmod +x your_package/your_python_script_dir/your_script.py
```

**Details:**

When VSCode builds your package it does so with 
```
colcon build --symlink-install --event-handlers console_cohesion+ --base-paths $your_path/ros_ws --cmake-args -DCMAKE_BUILD_TYPE=RelWithDebInfo
```

hence with flag `--symlink-install` which essentially ensures that the python-script you run (in `install/share/...`) points to the python script in your `src`-directory. When the python script in your `src` is not executable, the debugger throws the error above. Hope it helps :)