ms-iot/vscode-ros #1300: [bug]

Achllle, 2024-09-19:
- [x] Linux:  Ubuntu 20.04
- [x] ROS 1: Noetic

version: main (35f2f3b9de60586eaee944f8457f5647dbb44354)
I'm running the extension in a devcontainer.

# what is the bug

When compiling ROS 1 packages using colcon instead of catkin_make, the `launch` debug task does not find the nodes being launched as the `workspaceOverlayPath` is incorrectly set to `devel` instead of `install`, causing an error dialog to come up with a 'ResourceNotFoundeError`. The offending logic for that is [here](https://github.com/ms-iot/vscode-ros/blob/35f2f3b9de60586eaee944f8457f5647dbb44354/src/extension.ts#L414-L423).

```typescript
    if (newEnv.ROS_VERSION === "1") {
        workspaceOverlayPath = path.join(`${vscode.workspace.rootPath}`, "devel_isolated");
        if (!await pfs.exists(workspaceOverlayPath)) {
            workspaceOverlayPath = path.join(`${vscode.workspace.rootPath}`, "devel");
        }
    } else {    // FUTURE: Revisit if ROS_VERSION changes - not clear it will be called 3
        if (!await pfs.exists(workspaceOverlayPath)) {
            workspaceOverlayPath = path.join(`${vscode.workspace.rootPath}`, "install");
        }
    }
```

## Repro steps

1. compile package with c++ code with colcon
2. create a launch.json task with request `launch` which launches a launch file that starts the c++ ROS node
3. set breakpoints in file
4. launch that debug job

# expected behavior

Debugger launches, finds node, launches it, stops at breakpoint

# additional context

I tried debugging this by using [the VSIX](https://code.visualstudio.com/api/advanced-topics/remote-extensions#installing-a-development-version-of-your-extension) steps since I'm running inside the container (took a while to figure this out, a link to this in the contributing.md file would have helped)

A simple fix could be to add another nested if statement:

```typescript
        if (!await pfs.exists(workspaceOverlayPath)) {
            workspaceOverlayPath = path.join(`${vscode.workspace.rootPath}`, "install");
```

though the better solution would be to not use the `ROS_VERSION` to inform the location but instead find whatever install or devel files exist.

VScode also tells me that `vscode.workspace.rootPath` is deprecated and `vscode.workspace.workspaceFolder` should be used.

Even with these fixes I couldn't get the debugger to actually break at the break points, they simply grey out, so I can't fully test the whole set up, but I'd be happy to create a PR.

Achllle, 2025-01-08:
@ooeygui This error also disappears on version 0.8.4, meaning this is a regression. Still get the greyed out breakpoints. This extension could *really* use some TLC from a maintainer - is this still being supported?

ooeygui, 2025-01-09:
@Achllle I'm in the process of migrating this extension outside of Microsoft (with permission), I'll be making announcement shortly.

One thing I'm doing in the process is splitting ROS 1 and ROS 2 into two different extensions; so they can be maintained independently.